ophub/amlogic-s9xxx-openwrt #818: 大佬是否可以加入荣品rp-rk3568-b

lakalin, 2025-09-17:
[image]
这是官方全套资料，我也不知道该发什么出来有用，大佬麻烦有空看看谢谢
RP-rk3568 链接：https://pan.baidu.com/s/1Z2fG_8VGUrYZiw1UqgSHwA 提取码：2345

lakalin, 2025-09-18:
@ophub 大佬有时间麻烦看看能不能支持，谢谢

lakalin, 2025-09-19:
好的麻烦大佬，我先试试

ophub, 2025-09-20:
选择使用这个：https://github.com/ophub/amlogic-s9xxx-armbian/pull/2736

miniloader下载这个：https://github.com/ophub/u-boot/tree/main/u-boot/rockchip/rp-rk3568

img从release里面下载

lakalin, 2025-09-20:
感谢大佬，我也在积极差这个东西的型号，刚刚又试了oec的这个miniloadre这个文件，瑞芯微的这个工具用自己下载的执行完第一个到img传输的时候就会立马闪退，

[image]
如果用官方给的就没有image这个分区，我尝试刷到userdate这个分区结果是不成功的

[image]

lakalin, 2025-09-20:
大佬我目前自己看了一下荣品的板子sdk都是根据他现在更新的镜像来的，这个对咱们armbian兼容是否有影响，每个机器都是根据三个系统给了三个sdk

[image]

[image]

ophub, 2025-09-21:
让你看oect的教程，是让你看miniloader和armbian.img的选择和地址的写法，不是让你用oect的miniloader和img。用我发给你的给你的设备编译的miniloader和img

lakalin, 2025-09-21:
谢谢大佬提醒，我用https://github.com/ophub/u-boot/tree/main/u-boot/rockchip/rp-rk3568这个地址的miniloader后还是同样的情况卡开机界面,我水平不高大佬见谅哈

[image]

lakalin, 2025-09-21:
sd卡是我烧录了一个官方的系统尝试从sd卡启动刷机的时候忘记取下来了，不过也同样没有成功

lakalin, 2025-09-21:
大佬，我用ttl跑码记录了启动的代码，然后用ai分析了固件写入后失败的原因，辛苦大佬有时间帮忙看看，谢谢

[image]
▒DDR fdeec6f4fc typ 23/09/25-19:41:25,fwver: v1.19
In
LP4/4x derate en, other dram:1x trefi
ddrconfig:7
LPDDR4, 324MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
tdqss: cs0 dqs0: -24ps, dqs1: -96ps, dqs2: -48ps, dqs3: -96ps,
tdqss: cs1 dqs0: -48ps, dqs1: -96ps, dqs2: -48ps, dqs3: -96ps,

change to: 324MHz
clk skew:0x62

change to: 528MHz
clk skew:0x58

change to: 780MHz
clk skew:0x58

change to: 1332MHz(final freq)
PHY drv:clk:38,ca:38,DQ:30,odt:60
vrefinner:16%, vrefout:29%
dram drv:40,odt:80
vref_ca:00000068
clk skew:0x22
cs 0:
the read training result:
DQS0:0x32, DQS1:0x31, DQS2:0x38, DQS3:0x37,
min  : 0x6  0x8  0x9  0x4  0x2  0x2  0x2  0x3 , 0x6  0x2  0x6  0x6  0x3  0x9  0x9  0x3 ,
       0x8  0x3  0x9  0x9  0x5  0xa  0xc  0x7 , 0x7  0x5  0x4  0x2  0x5  0x5  0x9  0x9 ,
mid  :0x20 0x23 0x24 0x20 0x1e 0x1f 0x1e 0x20 ,0x21 0x1e 0x21 0x22 0x1e 0x24 0x24 0x20 ,
      0x22 0x1e 0x23 0x22 0x21 0x25 0x25 0x22 ,0x23 0x20 0x20 0x1e 0x21 0x21 0x25 0x25 ,
max  :0x3b 0x3f 0x3f 0x3d 0x3b 0x3c 0x3a 0x3d ,0x3d 0x3b 0x3d 0x3e 0x3a 0x3f 0x40 0x3d ,
      0x3d 0x39 0x3d 0x3c 0x3d 0x41 0x3f 0x3e ,0x3f 0x3c 0x3d 0x3a 0x3d 0x3e 0x41 0x41 ,
range:0x35 0x37 0x36 0x39 0x39 0x3a 0x38 0x3a ,0x37 0x39 0x37 0x38 0x37 0x36 0x37 0x3a ,
      0x35 0x36 0x34 0x33 0x38 0x37 0x33 0x37 ,0x38 0x37 0x39 0x38 0x38 0x39 0x38 0x38 ,
the write training result:
DQS0:0x1e, DQS1:0x12, DQS2:0x1a, DQS3:0x12,
min  :0x52 0x55 0x57 0x54 0x50 0x50 0x4f 0x53 0x54 ,0x49 0x45 0x49 0x49 0x44 0x4a 0x4d 0x46 0x48 ,
      0x49 0x48 0x4b 0x4b 0x48 0x4c 0x4c 0x4b 0x4a ,0x44 0x43 0x42 0x3e 0x43 0x3f 0x44 0x47 0x45 ,
mid  :0x6f 0x71 0x73 0x70 0x6b 0x6b 0x6b 0x6f 0x70 ,0x66 0x61 0x65 0x65 0x5e 0x66 0x67 0x61 0x64 ,
      0x66 0x65 0x68 0x67 0x64 0x68 0x67 0x66 0x66 ,0x60 0x5f 0x5e 0x5b 0x5e 0x5c 0x5f 0x64 0x5f ,
max  :0x8c 0x8d 0x8f 0x8c 0x86 0x86 0x88 0x8b 0x8c ,0x84 0x7e 0x81 0x82 0x79 0x82 0x82 0x7d 0x81 ,
      0x84 0x82 0x85 0x84 0x80 0x85 0x82 0x81 0x82 ,0x7c 0x7b 0x7a 0x78 0x79 0x79 0x7a 0x81 0x79 ,
range:0x3a 0x38 0x38 0x38 0x36 0x36 0x39 0x38 0x38 ,0x3b 0x39 0x38 0x39 0x35 0x38 0x35 0x37 0x39 ,
      0x3b 0x3a 0x3a 0x39 0x38 0x39 0x36 0x36 0x38 ,0x38 0x38 0x38 0x3a 0x36 0x3a 0x36 0x3a 0x34 ,
cs 1:
the read training result:
DQS0:0x2e, DQS1:0x2e, DQS2:0x38, DQS3:0x38,
min  : 0x6  0x7  0x8  0x4  0x2  0x2  0x2  0x3 , 0x6  0x2  0x6  0x7  0x3  0x9  0x9  0x3 ,
       0x6  0x2  0x7  0x7  0x4  0x9  0x9  0x6 , 0x9  0x6  0x5  0x2  0x5  0x6  0x9  0xa ,
mid  :0x20 0x21 0x22 0x1f 0x1d 0x1d 0x1c 0x1e ,0x21 0x1d 0x20 0x21 0x1d 0x23 0x24 0x1e ,
      0x21 0x1d 0x21 0x21 0x1f 0x24 0x23 0x21 ,0x25 0x22 0x21 0x1e 0x22 0x22 0x25 0x26 ,
max  :0x3a 0x3b 0x3d 0x3a 0x39 0x39 0x37 0x3a ,0x3d 0x39 0x3b 0x3b 0x37 0x3d 0x3f 0x3a ,
      0x3d 0x39 0x3c 0x3b 0x3b 0x40 0x3e 0x3d ,0x41 0x3e 0x3e 0x3a 0x3f 0x3f 0x41 0x43 ,
range:0x34 0x34 0x35 0x36 0x37 0x37 0x35 0x37 ,0x37 0x37 0x35 0x34 0x34 0x34 0x36 0x37 ,
      0x37 0x37 0x35 0x34 0x37 0x37 0x35 0x37 ,0x38 0x38 0x39 0x38 0x3a 0x39 0x38 0x39 ,
the write training result:
DQS0:0x1e, DQS1:0x12, DQS2:0x1a, DQS3:0x12,
min  :0x50 0x54 0x55 0x51 0x4f 0x4f 0x4d 0x51 0x53 ,0x4b 0x48 0x4c 0x4b 0x48 0x4d 0x4f 0x49 0x4b ,
      0x4c 0x4b 0x50 0x50 0x4c 0x51 0x50 0x50 0x4e ,0x49 0x47 0x47 0x45 0x48 0x47 0x49 0x4d 0x49 ,
mid  :0x6d 0x6f 0x71 0x6d 0x6a 0x6a 0x69 0x6d 0x6f ,0x68 0x65 0x68 0x68 0x62 0x68 0x6a 0x65 0x67 ,
      0x6a 0x68 0x6c 0x6c 0x68 0x6d 0x6b 0x6b 0x6a ,0x66 0x64 0x64 0x61 0x64 0x62 0x65 0x69 0x64 ,
max  :0x8b 0x8b 0x8e 0x8a 0x86 0x85 0x86 0x89 0x8b ,0x86 0x82 0x85 0x85 0x7c 0x84 0x86 0x81 0x83 ,
      0x88 0x86 0x89 0x88 0x85 0x89 0x87 0x86 0x86 ,0x83 0x81 0x81 0x7d 0x81 0x7d 0x82 0x86 0x80 ,
range:0x3b 0x37 0x39 0x39 0x37 0x36 0x39 0x38 0x38 ,0x3b 0x3a 0x39 0x3a 0x34 0x37 0x37 0x38 0x38 ,
      0x3c 0x3b 0x39 0x38 0x39 0x38 0x37 0x36 0x38 ,0x3a 0x3a 0x3a 0x38 0x39 0x36 0x39 0x39 0x37 ,
CA Training result:
[... 1382 lines cut ...]
[   11.338307]  92:          0          0          0          0  rockchip_gpio_irq   3 Level     rk817
[   11.347339]  93:          0          0          0          0     rk817   0 Edge      rk805_pwrkey_fall
[   11.356635]  94:          0          0          0          0     rk817   1 Edge      rk805_pwrkey_rise
[   11.365932]  98:          0          0          0          0     rk817   5 Edge      RTC alarm
[   11.374534] 117:          0          0          0          0  rockchip_gpio_irq  13 Edge      bt_default_wake_host_irq
[   11.385216] 118:          0          0          0          0  rockchip_gpio_irq  18 Edge      headset_input
[   11.394944] PMU:
[   11.396787] pwr       0x00a0: 00000077 00000000 00000003 00000000
[   11.402872] status    0x0098: 00000077 00000000 00000077 00000000
[   11.408957] req       0x0050: 0000016e 00000000 00000000 00000000
[   11.415041] idle      0x0068: 0000016e 00000000 0000ffff 0000000f
[   11.421126] ack       0x0060: 0000016e 00000000 0000016e 00000000
[   11.427214] rockchip-thermal fe710000.tsadc: channal 0: temperature(25 C)
[   11.433993] THERMAL REGS:
[   11.436612] 00000000: 00000fc0 00030033 00000301 00000002 00000000 00000000 00000000 00000000
[   11.445129] 00000020: 000007f4 000007f8 00000000 00000000 00000940 00000000 00000000 00000000
[   11.453645] 00000040: 00000a70 00000a70 00000000 00000000 00000000 00000000 00000000 00000000
[   11.462162] 00000060: 00000004 00000004 00000656 00000656 00000000 00000000 00000000 00000000
[   11.470675] 00000080: 00000000 00000000
[   11.474504] Kernel Offset: disabled
[   11.477985] CPU features: 0x000,08000026,2a00aa38
[   11.482681] Memory Limit: none
[   11.485737] cpu cpu0: cur_freq: 1992000000 Hz, volt: 1150000 uV
[   11.503333] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,6) ]---
[emmc启动paoma.txt](https://github.com/user-attachments/files/22452365/emmc.paoma.txt)

ophub, 2025-09-22:
```
[    5.163084] No filesystem could mount root, tried: ext3 ext2 ext4 squashfs vfat exfat ...
[    5.163130] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,6)
[    5.162969] List of all partitions:
[    5.162997] 0100            4096 ram0
[    5.162999]  (driver?)
[    5.163014] b300         7634944 mmcblk0
[    5.163016]  driver: mmcblk
[    5.163026]   b301            4096 mmcblk0p1 696e0000-0000-4331-8000-7c4c000022ad
[    5.163028]
[    5.163036]   b302            4096 mmcblk0p2 ac200000-0000-4b6a-8000-2a3c000009b4
[    5.163038]
[    5.163046]   b303           65536 mmcblk0p3 f9490000-0000-4540-8000-2879000020e1
[    5.163048]
[    5.163056]   b304          131072 mmcblk0p4 d8380000-0000-4b61-8000-678a0000073b
[    5.163057]
[    5.163065]   b305           32768 mmcblk0p5 d1030000-0000-4500-8000-1ff7000060ca
[    5.163066]
[    5.163074]   b306         7389167 mmcblk0p6 614e0000-0000-4b53-8000-1d28000054a9
```

帅哥，地址不对，地址不对，你仔细看oect的刷机教程里的地址，图片和文字里我都写了地址。
你没有写对地址所以之前你emmc上刷入的其他系统还在，从你的启动日志看还是6个分区，这是你之前的系统。现在的armbian只有2个分区。

ophub, 2025-09-22:
Miniloader.bin 这个文件，你之前刷官方固件使用的是哪个，可以继续使用官方的，这是引导文件。
你之前刷官方的ubuntu/debian系统选择的是哪个镜像？刷完官方系统进去后看他使用了哪个dtb文件，也可以推断出是哪个编译选项。

ophub, 2025-09-22:
在SDK里没找到这个，我先回退仓库移除这个设备的添加了。等确认了版本再试。

closed 2025-10-29 by ophub

lakalin, 2025-11-10:
> > 你测试的就是49编译的
> 
> 大佬，我这边提取了官方系统上面的dtb文件然后转换了dts文件是否可以直接重新打包呢？
> 
> [dtbdts.zip](https://github.com/user-attachments/files/23248827/dtbdts.zip)

@ophub 麻烦大佬看一眼提取的dtb和dts是否可以使用嘞